OceanDataTools/openrvdas #463: Logger_manager hangs after updating configs or turning off loggers

sheehanigans-at-work, 2025-09-16:
Not sure if anyone else has come across this issue, but every now and then our logger_manager will get stuck and needs a restart from supervisor. Usually happens when updating configs, but can happen when switching logging off/on again. It can be a bit tricky to replicate as sometimes it happens quickly and other times not. But to replicate I generally keep turning all the loggers off and on again.

I've tracked it down to the quit() function in the logger_runner.py. From what I can see, a process gets stuck and isn’t terminating (not sure why), and because of the self.process.join() it hangs indefinitely until the logger_manager is restarted.
 

[image]

[image]

 
I did experiment and added a timeout of 5 seconds to the process.join(), and if it timed out I would use os.kill to kill the process, and things wouldn’t seem to hang after that.

Thoughts on this issue? Anyone else have it happen to them or able to replicate? Best solution?

I'm using
OS: Ubuntu 22.04
Python: 3.10.12

Cheers,
Kieran

sheehanigans-at-work, 2025-09-16:
Hi Webb,

Yes, I believe we have pulled in the latest changes into our repo. We had this issue for a while now but only have just had a chance to start investigating it. 

I might just try and spin up a completely fresh instance without any of our stuff, still see if I can recreate it there.

davidpablocohn, 2025-09-17:
Thanks for this, Kieran!

I've not had other reports of this behavior, but I certainly wouldn't rule
it out. My experience is that a lot of folks just jury rig something when
it doesn't work as expected, and never file a bug report. Regardless, I
like your solution of a timeout + os.kill; I can't see a lot of downside to
adding that, and will have a look at adding it to the dev branch.

Again, thanks!

davidpablocohn, 2025-09-18:
Hi Kieran - could you have a look at
https://github.com/OceanDataTools/openrvdas/pull/464 ?

It's an implementation of the fix you've described working for you.

PR #464 (2025-09-22): Resolves #463 - try terminating process gently, and escalate if failing.
Resolves #463
closed 2025-09-22 by davidpablocohn